nand2tetris/web-ide #554: [bug]: CPU EMULATOR FOR ONLINE IDE NOT ACCEPTING ENTRY OF VARIABLES

mikelexx, 2025-01-08:
### Tool

CPU Emulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

murithimichael254@gmail.com

### What happened?

A bug happened!I expected if I type '@variable' and  press enter, the ROM part should write @variable, but it found its clearing the input field to he original '@0'

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

h0dgep0dge, 2025-02-01:
This is intended behaviour, the CPU emulator doesn't handle variables, you'll want to put your code through the assembler first.

DavidSouther, 2025-08-10:
Please use the Assembler tool, then click the "Load in CPU Emulator" button.

closed 2025-08-10 by DavidSouther