nand2tetris/web-ide #556: [bug]: Comparison failure: line 0, for identical lines !

pierrotseban, 2025-01-10:
### Tool

Assembler

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

pierrotseban@gmail.com

### What happened?

When I try to compare my hack code with the one generated by the assembler, I get "comparison failure: line 0", but both lines 0 are all 0s. I feel dumb but the behavior is very weird and I did not find anything online or in the guide.

[image]

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maueroats, 2025-01-11:
Please include enough information for someone else to reproduce the bug!

That means provide binary code from the right side.

This is from `projects/06/Rect.asm`. (Stating this in the bug report makes it clearer.)

Unable to reproduce with WebIDE Version 2024.52.0 (which presumably is the one you also used).

I verified that using dos line endings is _not_ the source of this bug.

pierrotseban, 2025-01-12:
Sorry ! 

It is WebIDE Version 2024.52.0, on Google Chrome, the code from the left side does come from Rect.asm, and here is my hack file

[Rect.zip](https://github.com/user-attachments/files/18388941/Rect.zip)

I reproduce if I just type 0000000000000000 etc. in a basic text editor and save it as a hack file (here joined).

[test.zip](https://github.com/user-attachments/files/18388955/test.zip)

maueroats, 2025-01-12:
Writing bug reports is hard - Did you verify that this exact file causes the error you see?  The file you provided on my computer has 25 lines of zeros but the output of Rect.asm has only 24 lines, so I get "Comparison failed - different lengths".

Removing the extra line of zeros but _not changing the dos line endings_ (0xa 0xd) makes this reproducible (Linux, Firefox).

maueroats, 2025-01-12:
Are you working on a Windows computer? As a temporary fix, try having the file you are uploading use only linefeeds and not CR LF. 

If you're not familiar with the issue, here are my quick search results: [convert on windows](https://stackoverflow.com/questions/17579553/windows-command-to-convert-unix-line-endings), [convert in a unix-centric environment](https://stackoverflow.com/questions/3891076/how-to-convert-windows-end-of-line-in-unix-end-of-line-cr-lf-to-lf).

Please post here if it works even if this isn't really a "fix".