nand2tetris/web-ide #561: [bug]: Test of Builtin Memory chip hanging in Chrome

kebasye, 2025-02-07:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

kebasye@clarku.edu

### What happened?

Working on Chrome under Mac/Sonoma, I consistently hang the browser running the standard test for the Memory chip.  This happens at the keyboard test, somewhere in this section:

set address 24576,
echo "Click the Keyboard icon and hold down the 'K' key (uppercase) until you see the next message...",
// It's important to keep holding the key down since if the system is busy,
// the memory will zero itself before being outputted.

while out <> 75 {
    tick, tock,     // tick, tock prevents hang if sync. parts used in KB path.
}

clear-echo,
output;

The echo doesn't show up in the interface and the rest of the interface becomes non-responsive (can't enable/disable keyboard, pause/reset test script, scroll, etc.). Pressing K has no effect.  Eventually, Chrome puts up the "page not responsive" dialog with options "Wait" and "Exit page."  Waiting doesn't help.  

When I replaced the comma at the end of the echo with a semicolon, the echo did show up when I stepped through the test.  The next step, which seems to be the while loop, clear-echo, and output, hung as before.  I had the keyboard enabled and was holding down the K key, but it was still hung.

Chrome: Version 133.0.6943.54 (Official Build) (arm64)
Mac: M2 running Sonoma - 14.1.1 (23B81)

### Additional Comments

Thanks for a terrific tool!

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

e-rairigh, 2025-07-21:
What does it mean "Click the Keyboard icon and hold down the 'K' key (uppercase) until you see the next message..."?

I can't find a keyboard icon.

DavidSouther, 2026-10-25:
Fixed in nand2tetris/projects#12, which uses `tick, tock;` in both keyboard loops of `Memory.tst`. The web IDE's built-in copy already did. The distributed zip will pick up the change in the next projects release. Thanks @Timzka for the diagnosis.

closed 2026-10-25 by DavidSouther