nand2tetris/web-ide #562: [bug]: Project 2 - ALU - negation (bitwise not) instructions are unclear

Zhomart, 2025-02-08:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

Here's the default instructions when Project 2 / ALU is loaded. It says when "nx == 1", it should do "bitwise not, x = !x" operation.
But in "CHIP ALU", "nx" says "negate the x input" - which is "x = -x"  (i.e. x = 1 + 2^n - x from the course video).

Can you please clarify if it should be "negative x" or "not x"?

```
// Implementation: Manipulates the x and y inputs
// and operates on the resulting values, as follows:
// if (zx == 1) sets x = 0        // 16-bit constant
// if (nx == 1) sets x = !x       // bitwise not
// if (zy == 1) sets y = 0        // 16-bit constant
// if (ny == 1) sets y = !y       // bitwise not
// if (f == 1)  sets out = x + y  // integer 2's complement addition
// if (f == 0)  sets out = x & y  // bitwise and
// if (no == 1) sets out = !out   // bitwise not

CHIP ALU {
    IN  
        x[16], y[16],  // 16-bit inputs        
        zx, // zero the x input?
        nx, // negate the x input?
        zy, // zero the y input?
        ny, // negate the y input?
        f,  // compute (out = x + y) or (out = x & y)?
        no; // negate the out output?
```

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Zhomart, 2025-02-08:
I got confused, "negate" means "bitwise not" and I was doing "-x".

I'll keep it this issue open, I think the instructions should be updated.

DavidSouther, 2025-08-10:
These instructions come directly from the book, Fig 2.5c

[image]

closed 2025-08-10 by DavidSouther