nand2tetris/web-ide #565: [bug]: Project 4 - Web IDE - CPU Emulator- Running Mult.tst and Fill.tst deletes user ROM input

FlorianMiedtke, 2025-03-02:
### Tool

CPU Emulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

When I want to run the test files for project 4, they delete the contents of the ROM. This seems to happen because of the line 10 of Mult.tst

https://github.com/nand2tetris/web-ide/blob/cd2963f4ad14c1f8ce99ae0f11d5a749cc28bc0c/projects/src/project_04/01_mult.ts#L21

This seems to load the default Mult.asm file of the Web IDE file system, which is always the empty default. An easy fix would be to just comment out this line, but I think this would ignore the underlying problem. 

If you look at the hardware simulator, the code written there is saved to the Web IDEs file system. So when I reload the page, my changes are still there. In the CPU Emulator, the ROM is also cleared, when I reload the page. So the actual problem seems to be, that the user input in the ROM is not saved to the Web IDEs file system.

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

amama22, 2025-03-09:
having the same problem. how did u manage to test it?

FlorianMiedtke, 2025-03-10:
> having the same problem. how did u manage to test it?

I just executed the defined test steps manually.

jimmyvh, 2025-03-23:
Just ran into this issue myself, and found this bug after googling.
As a workaround, executing the fist step first, then filling out the asm code works, but it's an annoying workaround.
The real issue is that the asm code is not being saved. 
Test failing and you need to make fixes? Fill in everything from scratch. Want to to let it sink into your brain for a while and come back later? Start from scratch. Could use an external text editor to save inbetween ofcourse, but you still have to manually type over everything between each iteration.
If the file actually gets saved like it does for the hardware simulator, everything would work as intended, the *load Mult.asm* line included.

gacarrillor, 2025-03-23:
Same issue here.

Duplicate of #551 and #543

amama22, 2025-03-23:
> Same issue here.
> 
> Duplicate of [#551](https://github.com/nand2tetris/web-ide/issues/551) and [#543](https://github.com/nand2tetris/web-ide/issues/543)

Hey, if this helps, I realized after, in the assembler u have to load the asm file lol and fill that, and then when u load that file in the cpu emulator it will correctly open that same file.

chids04, 2025-07-24:
can confirm this issue is still occuring

commit 5d44c798 (2025-08-22): Allow users to upload local .tst and .cmp files in CPU Emulator (#591)

* Allow users to upload local .tst and .cmp files in CPU Emulator

Fixes #543
Fixes #565
Fixes #551
Fixes #543

---------
closed 2025-08-22 by DavidSouther