nand2tetris/web-ide #568: [bug]: Project 4, Mult.tst file contains wrong RAM[2] initialization value

xprose7820, 2025-03-10:
### Tool

None

### Interface

None

### What happened?

should be set to ----> RAM[2] = 0, not RAM[2] = -1 

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

v3ai, 2025-04-01:
Think it's checking that you init it to zero and don't assume it to be zero

DavidSouther, 2025-08-10:
This is correctly verifying your program sets RAM[2] to 9, as stated in the comment in the test file:

[image]