nand2tetris/web-ide #574: [bug]: well, web based assembler is unusable

alex1yaremchuk, 2025-04-24:
### Tool

Assembler

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

sanyaya@bk.ru

### What happened?

Hi, I'm very disappointed in assembler web version.

1) assembler editor with command-by-command input is completely unusable. 
click the cell and then input the command - really?
why not just regular text editor multiline?

2) fine.. I made my file locally in vs code and tried to run tests on it in web assembler. 
no - not possible - you can load file but not tests or tests but not file. 
it's just insane. 

and the site officially advices to use web version.. 
wow, I'm happy I have done it with nandgame first or I would hate the idea completely at this step.

### Additional Comments

Thank you anyways, the whole project is mind-blowing.

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maueroats, 2025-07-09:
It sounds like you are using the CPU emulator, not the assembler. Use this: https://nand2tetris.github.io/web-ide/asm

But yeah, the 20 year old versions of everything work better. I prefer them.

https://github.com/nand2tetris/nand2tetris_simulator

alex1yaremchuk, 2025-07-10:
Thank you for your reply! 

I've now finished all the projects with emulators. 

next step is "simulate" it in FPGA. :)

> It sounds like you are using the CPU emulator, not the assembler. Use this: https://nand2tetris.github.io/web-ide/asm
> 
> But yeah, the 20 year old versions of everything work better. I prefer them.
> 
> https://github.com/nand2tetris/nand2tetris_simulator

DavidSouther, 2025-08-10:
The Java desktop apps are still available and you're of course welcome to use them. The online version is easier for some students to get started quickly, albeit with some advanced things like local files more difficult given it's a web-based application.

closed 2025-08-10 by DavidSouther