nand2tetris/web-ide #575: [feature]: Add Or16Way

tom-mai-wichtig, 2025-04-26:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

talatom154@gmail.com

### What feature are you proposing?

Add Or16Way in Project 1
In Project 2 when designing the ALU, the out zr = can be implemented with Or16Way
This cannot be done since we only have Or8Way and hdl doesn't support internal pin indexing to use Or8Way twice.
Using directly the Out also raise error: Cannot use output as input.

### Do you want to try to add this feature?

- [ ] I want to try to add this feature!

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

high-sc0re, 2025-06-01:
I was stuck on this too until I found bus-subsets (Slide 68 of lecture 2 slides)

DavidSouther, 2025-09-17:
It's unlikely we will add Or16way, as none of the projects specifically require it, and sub-bus or-8way is "good enough". That said, when #486 is implemented you would be able to add your own Or16way.